STMicroelectronics/fp-sns-datalog2 #3: Corrupted data in long measurements

nicolasj92, 2023-08-14:
We are using the STEVAL-STWINBX1 at a CNC machine to gather vibration data. We are using the fp-sns-datalog2 HSDatalog Python framework to record the data over USB. Unfortunately, the data seems to get corrupted in longer measurements (>5 min).

In these cases, the dataframes get corrupted and it seems as though the axis of the accelerometer are switched around

The following code is used to read the raw measurement files:

    from st_hsdatalog.HSD.HSDatalog import HSDatalog
    hsd_factory = HSDatalog()
    hsd = hsd_factory.create_hsd(acq_folder)
    
    sensor = HSDatalog.get_sensor(hsd, sensor_name)
    stwin_data = HSDatalog.get_dataframe(hsd, sensor)[0]

and the resulting dataframe shows the corrupt data - in the beginning, everything is ok but in the end, the timestamps are invalid and it seems as though the axis are switched:

[image]

The HSDatalog call gives the following warnings:
`
dtmi found in locally in base supported models
dtmi: dtmi/appconfig/steval_stwinbx1/fpSnsDatalog2_datalog2-2.json
[c:\Users\N.Jourdan_Lokal\anaconda3\lib\site-packages\numpy\core\function_base.py:128](file:///C:/Users/N.Jourdan_Lokal/anaconda3/lib/site-packages/numpy/core/function_base.py:128): RuntimeWarning: invalid value encountered in multiply
  stop  = asanyarray(stop)  * 1.0
[c:\Users\N.Jourdan_Lokal\anaconda3\lib\site-packages\numpy\core\function_base.py:127](file:///C:/Users/N.Jourdan_Lokal/anaconda3/lib/site-packages/numpy/core/function_base.py:127): RuntimeWarning: invalid value encountered in multiply
  start = asanyarray(start) * 1.0`
`

nicolasj92, 2023-08-14:
Additional info: We tried with 3 different hardware instances with the same results, so it has to be a software problem.

pradolis, 2023-08-23:
Let me report also here the comments we already exchanged on ST forum: [link](https://community.st.com/t5/mems-sensors/hsdatalog2-problem-with-stwinbx1/m-p/584389)

pradolis, 2023-08-23:
USB streaming is controlled by the PC. Host PC is the master, asking data from the board, so events like blocking the PC can impact dramatically on the acquisition. Other actions impacting acquisition performance could be high CPU usage on the PC or antivirus control on USB ports. We are already aware about this limitation, that DATALOG2 can't control. See Datalog troubleshooting chapter in [User manual.](https://www.st.com/resource/en/user_manual/um3106-getting-started-with-the-stm32cube-function-pack-for-high-speed-datalogging-and-ultrasound-processing-stmicroelectronics.pdf)

closed 2023-09-04 by pradolis

nicolasj92, 2023-09-05:
I don`t understand how this issue can be closed as **completed**. Yes the explanation is that there is some USB interference or something similar but why is all data from that moment on corrupted? This basically makes it impossible to take measurements for more than a few minutes. I would see it as a bug that the software cannot recover from a transmission problem that probably only lasts a few milliseconds.