STMicroelectronics/fp-sns-datalog2 #4: Python hsdatalog_cli.py does not work with provided configs

nicolasj92, 2023-08-15:
When trying to start the hsdatalog_cli.py in Utilities\HSDPython_SDK\examples\hsdatalog_cli.py with any of the provided example configs in Utilities\STWIN.box_config_examples the script crashes with an error on startup.

Steps to reproduce:
`python fp-sns-datalog2\Utilities\HSDPython_SDK\examples\hsdatalog_cli.py -f fp-sns-datalog2\Utilities\STWIN.box_config_examples\all_sensors.json`

The problem seems to be that dictionaries are used for the "samples_per_ts" and this is not accepted by PnPLHSD_com_manager.py as it only accepts bool int float or str properties:

  File "C:\Users\N.Jourdan_Lokal\projects\fp-sns-datalog2\Utilities\HSDPython_SDK\examples\Views\cli_views.py", line 409, in _update
    self._hsd_info_model.upload_device_conf_file()
  File "C:\Users\N.Jourdan_Lokal\projects\fp-sns-datalog2\Utilities\HSDPython_SDK\examples\hsdatalog_cli.py", line 143, in upload_device_conf_file
    res = HSDLink.update_device(self.hsd_link, self.selected_device_id, self.cli_flags.file_config)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\N.Jourdan_Lokal\anaconda3\Lib\site-packages\st_hsdatalog\HSD_link\HSDLink.py", line 422, in update_device
    return hsd_link.update_device(device_id, device_config_file)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\N.Jourdan_Lokal\anaconda3\Lib\site-packages\st_hsdatalog\HSD_link\HSDLink_v2.py", line 407, in update_device
    return self.__com_manager.update_device(d_id, device_json_file_path, self.__dt_manager.get_components())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\N.Jourdan_Lokal\anaconda3\Lib\site-packages\st_hsdatalog\HSD_link\communication\PnPL_HSD\PnPLHSD_com_manager.py", line 389, in update_device
    self.__set_component_values(d_id, component, dtdl_components[key])
  File "C:\Users\N.Jourdan_Lokal\anaconda3\Lib\site-packages\st_hsdatalog\HSD_link\communication\PnPL_HSD\PnPLHSD_com_manager.py", line 371, in __set_component_values
    self.set_property(d_id, comp[comp_name][content.name], comp_name, content.name)
  File "C:\Users\N.Jourdan_Lokal\anaconda3\Lib\site-packages\st_hsdatalog\HSD_link\communication\PnPL_HSD\PnPLHSD_com_manager.py", line 344, in set_property
    print(res)
          ^^^
UnboundLocalError: cannot access local variable 'res' where it is not associated with a value

pradolis, 2023-08-23:
Issue noted. Will be addressed in v1.3.0 in plan for wk40

pradolis, 2023-10-11:
Hello @nicolasj92
we've just pushed v2.0.0, that includes proper json config files.
I'm closing the issue

closed 2023-10-11 by pradolis