STMicroelectronics/fp-sns-datalog2 #5: Ultrasound FFT does not start as expected

smellit, 2023-08-29:
I cloned the repository and set up Python (V3.10.11) to uses the HSDPyhtonSDK together with STEVAL-STWINBX1. The application hsdatalog_GUI works well. But when I start the ultrasound_fft_app, The application shows the GUI as expected, but when I connect, the GUI view is not the expected one, but the same as hsdatalog_GUI, see here below.

[image]

[image]

Here the logfile: [20230829_07_54_37_app_debug.log](https://github.com/STMicroelectronics/fp-sns-datalog2/files/12460806/20230829_07_54_37_app_debug.log)

Here the terminal output:

```
No HSDatalog_v1 devices connected!
Trying to open communication using libhs_datalog_v2...
64bit
HSD_DLLv2: DeviceManager created
HSD_DLLv2: libusb_close! 
HSD_DLLv2: libusb_close! 
HSD_DLLv2: libusb_close! 
HSD_DLLv2: libusb_close! 
HSD_DLLv2: libusb_close! 

HSD_DLLv2: CONNECTED!    
Commmunication Opened correctly
qt.pysideplugin: Environment variable PYSIDE_DESIGNER_PLUGINS is not set, bailing out.
dtmi found in locally in base supported models
dtmi: dtmi/appconfig/steval_stwinbx1/fpSnsDatalog2_ultrasoundFft-3.json
dtmi found in locally in base supported models
dtmi: dtmi/appconfig/steval_stwinbx1/fpSnsDatalog2_datalog2-3.json     
2023-08-29 07:37:33,574 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: iis3dwb_acc - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component iis3dwb_acc Updated correctly
2023-08-29 07:37:33,685 - HSDatalogApp.st_hsdatalog.HSD_link.communication.PnPL_HSD.PnPLHSD_com_manager - ERROR - No Component Status[d_id:0,iis3dwb_ext_acc] returned. (PnPLHSD_com_manager.py:234)
2023-08-29 07:37:33,711 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_Controller - WARNING - The component [iis3dwb_ext_acc] defined in DeviceTemplate has not a Twin in Device Status from the FW (HSD_Controller.py:440)
ComponentWidget - WARNING - No status to update for iis3dwb_ext_acc Component
2023-08-29 07:37:33,879 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: iis2mdc_mag - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component iis2mdc_mag Updated correctly
2023-08-29 07:37:34,041 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: imp23absu_mic - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component imp23absu_mic Updated correctly
2023-08-29 07:37:34,209 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: ism330dhcx_acc - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component ism330dhcx_acc Updated correctly
2023-08-29 07:37:34,365 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: ism330dhcx_gyro - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component ism330dhcx_gyro Updated correctly
2023-08-29 07:37:34,548 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: ism330dhcx_mlc - status: False (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component ism330dhcx_mlc Updated correctly
2023-08-29 07:37:34,745 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: iis2dlpc_acc - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component iis2dlpc_acc Updated correctly
2023-08-29 07:37:34,904 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: stts22h_temp - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component stts22h_temp Updated correctly
2023-08-29 07:37:35,014 - HSDatalogApp.st_hsdatalog.HSD_link.communication.PnPL_HSD.PnPLHSD_com_manager - ERROR - Component Status[d_id:0,stts22h_ext_temp] parsing error. (PnPLHSD_com_manager.py:232)
2023-08-29 07:37:35,029 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - WARNING - It is impossible to know the Sensor [stts22h_ext_temp] enabling status from the FW device status (HSD_DeviceConfigPage.py:162)
2023-08-29 07:37:35,042 - HSDatalogApp.st_hsdatalog.HSD_link.communication.PnPL_HSD.PnPLHSD_com_manager - ERROR - Component Status[d_id:0,stts22h_ext_temp] parsing error. (PnPLHSD_com_manager.py:232)
2023-08-29 07:37:35,042 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_Controller - WARNING - The component [stts22h_ext_temp] defined in DeviceTemplate has not a Twin in Device Status from the FW (HSD_Controller.py:440)
ComponentWidget - WARNING - No status to update for stts22h_ext_temp Component
2023-08-29 07:37:35,186 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: ilps22qs_press - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component ilps22qs_press Updated correctly
2023-08-29 07:37:35,350 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: imp34dt05_mic - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component imp34dt05_mic Updated correctly
2023-08-29 07:37:35,520 - HSDatalogApp.st_hsdatalog.HSD_GUI.HSD_DeviceConfigPage - DEBUG - comp_name: iis2iclx_acc - status: True (HSD_DeviceConfigPage.py:157)
ComponentWidget - INFO - Component iis2iclx_acc Updated correctly
ComponentWidget - INFO - Component automode Updated correctly
ComponentWidget - INFO - Component log_controller Updated correctly
2023-08-29 07:37:35,829 - HSDatalogApp.st_hsdatalog.HSD_GUI.Widgets.TagsInfoWidget - DEBUG - Component: tags_info (TagsInfoWidget.py:186)
TagsInfoWidget - INFO - Component tags_info Updated correctly
ComponentWidget - WARNING - Property type not supported. (comp: acquisition_info, cont:tags) status not updated
ComponentWidget - INFO - Component acquisition_info Updated correctly
ComponentWidget - INFO - Component firmware_info Updated correctly
ComponentWidget - INFO - Component DeviceInformation Updated correctly

```

pradolis, 2023-08-29:
Hello @smellit ,
UltrasoundFFT needs the proper firmware to work as you are expecting.
Looking at the log you shared and at the screenshot from the GUI (where you can see the fw name "...FP-SNS-DATALOG2_Datalog2..."), I'm guessing you uploaded Datalog2 FW on the board. 
You must first flash the board with UltrasoundFFT example before using the UltrasoundFFT GUI.

Feel free to reopen the issue or create another one for any further support.

closed 2023-08-29 by pradolis

smellit, 2023-08-29:
Hello @pradolis 

Thanks for your help. I also found out my mistake and now I can connect correctly to the STEVAL-STWINBX1 and Ultrasound FFT works as expected.

:+1: