STMicroelectronics/fp-sns-datalog2 #8: HSDatalog and Python SDK installation not working on 64-bit ARM (Raspberry Pi 5)

nikitakuklev, 2023-12-07:
**Describe the set-up**
STEVAL-STWINBX1 + FP-SNS-DATALOG2 2.0.1
Raspberry Pi 5 8GB with Raspberry Pi OS 64-bit (latest)
Linux 6.1.65-v8-16k+ #1703 SMP PREEMPT Tue Dec  5 16:28:46 GMT 2023 aarch64

**Describe the bug**
Python installation and the underlying HSDatalog library do not work on 64-bit ARM. Using 64-bit is necessary to make use of all memory on Raspberry Pi 5 models.

**How To Reproduce the bug**
Try to compile https://github.com/STMicroelectronics/fp-sns-datalog2/blob/main/Utilities/cli_example/CMakeLists.txt (after fixing paths)

`
/usr/bin/c++  -std=c++11 -rdynamic CMakeFiles/cli_example.dir/main.cpp.o -o cli_example  -Wl,-rpath,<>/STM32CubeFunctionPack_DATALOG2_V2.0.1/Utilities/cli_example/lib/libhs_datalog/raspberryPi <>/STM32CubeFunctionPack_DATALOG2_V2.0.1/Utilities/cli_example/lib/libhs_datalog/raspberryPi/libhs_datalog_v2.so -lpthread 
`
`
/usr/bin/ld: <>/STM32CubeFunctionPack_DATALOG2_V2.0.1/Utilities/cli_example/lib/libhs_datalog/raspberryPi/libhs_datalog_v2.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
`

**Additional context/workarounds**

For Python, problem is that PySide 6.4.3 aarch64 is not on pip. Removing that requirement from the wheel allows installation to succeed.

For HSDatalog, either release the source code for libhs_datalog so that we can recompile ourselves, or provide aarch64 version

pradolis, 2023-12-11:
Hello @nikitakuklev ,
As for now, Python SDK and libhs_datalog has been developed and tested only up to Raspberry Pi 4 version running 32-bit OS only.
We've not already planned to include the support for Pi 5 / 64-bit OS on RPI, but we can work on it for the next version.

The package is including a 64-bit Linux version in 'libhs_datalog/linux' folder. It is not ARM based, so I'm not expecting it can be compatible with Pi 5, but that's something we've never tested.
If can you make an experiment and come back with your results, it can be very useful for us.

Best regards

pradolis, 2023-12-11:
@nikitakuklev let me add a further question: are you interested in using the Python SDK or would you use only the C++ based cli example?

To properly use the Python SDK, ready-to-use installers are available.
I suggest you to use HSDPython_SDK_install_noGUI script.
Be aware that PySide 6.4.3 is not available in any Raspberry environment and GUI example has never been supported in our SDK for RasPi. That's why we also included a "noGUI" version.

To use instead the C++ cli, you should be able to use the already complied files for RasPi, by following the instructions provided in the README files.
Are you able to launch the executable available in bin_RaspberryPi folder?
CMakeLists has been released only to recompile the cli_example executable only if you need to customize it.

Best regards

nikitakuklev, 2023-12-11:
@pradolis Thanks for getting back to me.

I have already tried libhs_datalog_v2 in linux folder on Raspberry Pi OS 64-bit, and it doesn't work as expected (need aarch64, not x86-64). Please add this to next version, because Raspberry Pi 5 cannot run in 32-bit mode at all - it is a hardware [limitation](https://forums.raspberrypi.com/viewtopic.php?t=360677)). Other people [requested this before](https://community.st.com/t5/stm32-mcus-products/dear-all-i-would-like-to-use-the-cli-example-utility-hs-datalog/td-p/230743), so hopefully things work out this time :)

Regarding Python vs C++, initially just need the HSDatalog SDK binary. We found a Raspberry Pi 4 with 32-bit OS and modified cli_example to log what we need - the goal is to replicate this setup on Raspberry Pi 5.

Regarding python GUI/installers, I modified Python wheels to use latest PySide6 (not 6.4.3) and GUI launched successfully on RasPi5 64-bit - all other packages were available. Of course, no devices were found, since SDK library fails to load on aarch64. Once aarch64 SDK is out, I can test the full functionality.

pradolis, 2023-12-12:
Hello @nikitakuklev 
Thanks for sharing your setup and also the ticket on community.st.com. We were not aware about it.
We'll work on RasPi 64-bit setup and add it in the next DATALOG2 version

pradolis, 2024-03-12:
Hello @nikitakuklev,
v2.1.0 has been released on [st.com](https://www.st.com/en/embedded-software/fp-sns-datalog2.html) only.
This new version includes the support for RasPi 64-bit, so I'm closing the issue.

Feel free to contact us again if the issue persists on your side.
Best regards

francus-dev, 2024-05-16:
> Hello @nikitakuklev, v2.1.0 has been released on [st.com](https://www.st.com/en/embedded-software/fp-sns-datalog2.html) only. This new version includes the support for RasPi 64-bit, so I'm closing the issue.
> 
> Feel free to contact us again if the issue persists on your side. Best regards

Hello @pradolis, just trying to use Datalog2_SDK with Raspberry Pi 4 Model B with Raspbian OS 64 bit. Your reply seems to suggest the cli_example in the folder "bin_raspberryPi4_64bit" of v.2.1.0 and 2.1.1 of the SDK should work correctly, but, after all has been correctly installed (without errors), trying to execute the file "cli_example", the following error appears: "error while loading shared libraries: libhs_datalog_v2.so: wrong ELFCLASS: ELFCLASS32" that makes me assume the library is for 32bit systems. How is it possible? Thanks in advance

pradolis, 2024-05-21:
Hello @francus-dev 
I've just reproduced your setup by using HSDPython_SDK from DATALOG2 v2.1.1 on Raspberry Pi 4 Model B with Raspbian OS 64 bit.
I'm running Debian 12 bookworm, 64 bit.
Can you check if you have the very same setup, by running those commands?
```
cat /etc/os-release
getconf LONG_BIT
```

There is a typo in the README_linux.md: we forgot to suggest to use the specific cli_example (from path bin_raspberryPi4_64bit). we are only referring to bin_linux folder. We can fix the readme instruction. Is this the possibile origin for your bug?

I launched bin_raspberryPi4_64bit/cli_example manually and it is working as expected.

Here the history of command I used to setup the environment:
```
cd cli_example/raspberryPi_setup
dos2unix 30-hsdatalog.rules setup.sh udev_removal.sh
chmod 777 setup.sh udev_removal.sh
./setup.sh
cd ../bin_raspberryPi4_64bit
chmod 777 cli_example
./cli_example
```

Best regards
Simone

Ferag-AG, 2024-06-20:
Hi @pradolis 

I also try to uses the Datalog2_SDK on a Raspberry Pi 5 and when I execute `cli_example` I also get following error as @francus-dev has mentioned.

`./cli_example: error while loading shared libraries: libhs_datalog_v2.so: wrong ELF class: ELFCLASS32`

Do I miss something to make it to work?

Here my configuration:

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
64
```

pradolis, 2024-06-20:
Hi all,
I'm reopening the issue.
DATALOG v2.2.0 will be released in a month, so we can add this check in pipe

Ferag-AG, 2024-06-20:
Hi @pradolis 

This will be great!

However I can confirm that Python based CLI as well as hsdatalog_GUI works on Pi5 and this is good for me.

alejandrolabajos, 2024-07-11:
Hello @Ferag-AG 

Could you give some information about how to run hsdatalog_GUI on RPI, please? 

Many thanks,
Alex

pradolis, 2024-07-16:
Hello,
I reshaped a bit the cli_example folder and the readme files in the upcoming v2.2.0.

Best regards
Simone

closed 2024-07-29 by pradolis