STMicroelectronics/fp-sns-datalog2 #9: DTDL2PnPL_cGen

smellit, 2023-12-12:
I'm adding a sensor and I wonder if it would be possible to get the DTDL2PnPL_cGen generator instead of having to implement a PnPL module by hand.

pradolis, 2024-01-29:
Hello @smellit 
sorry for the delay in the response.
We are still working on it. Soon we'll push it in a dedicated repo.
I let the issue opened and notify when and where the generator will be pushed.

Best regards
Simone

smellit, 2024-01-29:
Hi @pradolis 

Good news! I added an optical position sensor externally to the STWIN.box and had to implement most of it by hand, so it would be nice to be able to automate this. I can't wait to try it out!

Kind regards
Ludwig

pradolis, 2024-07-10:
Hello @smellit 
I'm sorry, but we have decided not to make the PnPL Generator externally available. I'm closing the issue, but I'll reopen and comment on it if the generator is published in the future.

Best regards
Simone

closed 2024-07-10 by pradolis