STMicroelectronics/fp-sns-datalog2 #10: Support for an optical position sensor.

Ferag-AG, 2024-02-02:
Dear ST Team

I am currently working on implementing a 2 dimensional position tracking sensor PAT9136 from PixArt in the FP-SNS-DATALOG2 firmware and I am asking me if this could be interresting for you to have this implementation merged. This would still have to be authorized by PixArt, as their technical data is confidential. I'll have PixArt visit us and ask for authorization if you're interested.

Regards,
Ludwig

pradolis, 2024-02-07:
Hello @Ferag-AG 
We would like to see your code.
If you agree, we can also continue here the discussion regarding the Python SDK support (as for your request on [community.st.com](https://community.st.com/t5/stm32-mcus-boards-and-hardware/extend-fp-sns-datalog2-phython-sdk-to-support-an-additional)).
Can you fork our repo and push there your developments?

Best regards
Simone

Ferag-AG, 2024-02-07:
Hi @pradolis

As the information provided by PixArt is classified as confidential, I must first obtain authorization before publishing my modifications. PixArt is going to visit me in march and I'll try to convince them that this could be an opportunity.

Best regards
Ludwig

Ferag-AG, 2024-04-29:
Hi

Unfortunately PixArt is to restricitve to allow me to publish the implementation, therfore this will not take place and the issue can be closed.

Best Regards
Ludwig

closed 2024-04-29 by Ferag-AG