STMicroelectronics/fp-sns-datalog2 #11: Battery measurement return voltage = 0

Ferag-AG, 2024-02-14:
Hi all

I am currently using the **fp-sns-datalog2** battery powered and I'm faced with a random effect.

While a SD card data recording is running, the recording stops to works after an irregular time interval. I go trough the steps which lead to this behaviour and I found out, that sometime the battery measurement returned by the function `UtilTask_GetBatteryStatus`, resp `BCPAcquireState` is equal to zero. This lead sending a message of type `DT_USER_BUTTON `in the function `DatalogAppTaskAdvOBTimerCallbackFunction` and stop data recording.

I use **fp-sns-datalog2** version 2.0.1 on the STWIN.box wiht the implementation of a custom sensor type whose data is read periodically (20 ms intervals) via the timer implemented in the corresponding task.

My IDE is Keil MDK uVision 5.38.0.0

I do not have taken a closer look to the battery management and my workaround is to discard battery values of zero in the `DatalogAppTaskAdvOBTimerCallbackFunction` 

Do you have a suspected reason for this behavior and a method to correct it?

Kind resgards,
Ludwig

pradolis, 2024-02-15:
Hello @Ferag-AG 
I'll try to reproduce your setup and check the possible issue.

Best regards
Simone

pradolis, 2024-02-15:
Hello @Ferag-AG 
I found the bug in the Battery Charger driver. 
I'm still not able to understand the reason, but as you said sometimes (and not regularly) the ADC used to calculate the voltage value from the battery fails.

As a valid workaround, you can modify this function in the BCAdcDriver.c file:
```
sys_error_code_t BCAdcDriver_GetValue(IDriver *_this, uint16_t *p_value)
{
  assert_param(_this != NULL);
  assert_param(p_value != NULL);
  sys_error_code_t res = SYS_NO_ERROR_CODE;
  BCAdcDriver_t *p_obj = (BCAdcDriver_t *) _this;
  ADC_HandleTypeDef *p_adc = p_obj->p_mx_adc_cfg->p_adc;
  uint32_t measured_voltage;

  if (HAL_ADC_PollForConversion(p_adc, 10) == HAL_OK)
  {
    measured_voltage = __LL_ADC_CALC_DATA_TO_VOLTAGE(p_adc->Instance, ADC_VREF, HAL_ADC_GetValue(p_adc), LL_ADC_RESOLUTION_12B);
    /* When HAL_ADC_GetValue fails, the returned value is 0 - workaround to avoid to use it */
    if (measured_voltage != 0)
    {
      *p_value = (uint16_t)((float)(measured_voltage) * ADC_BATMS_RATIO);
    }
    else
    {
      res = SYS_BASE_ERROR_CODE;
    }
  }
  else
  {
    res = SYS_BASE_ERROR_CODE;
  }

  return res;
}
```

I'll try to understand the root cause. In any case, at least the above workaround will be added in DATALOG2 v2.1.0 we are closing soon.

Let me know if this fix can solve the issue also on your side, so that I can close the issue.
Best regards
Simone

Ferag-AG, 2024-02-15:
Hi @pradolis 

Thanks for the confirmation.

I have implemented your workaround but there are still some condition where the measured battery voltage is too low which also triggered the data recording stop. I put a debug output in the function `UtilTask_GetBatteryStatus` and I get lots of lower values outputted.
```
UTIL task: Low battery voltage = 2223
UTIL task: Low battery voltage = 2705
UTIL task: Low battery voltage = 2134
UTIL task: Low battery voltage = 2603
UTIL task: Low battery voltage = 1951
UTIL task: Low battery voltage = 2428
```
So this workaround does not fix the issue.

I hope this can help!

Kind regards
Ludwig

Ferag-AG, 2024-02-16:
Hi @pradolis 

I also noticed an other behavior that may be related to this one.

In the sensor task that I have added to **fp-sns-datalog2** data are requested from an optical tracking sensor connected externally to the STWIN.box over the SPI 2 interface on a time interval way (every 20 ms) using the timer in the task. On timer expiration the function `...TaskTimerCallbackFunction` sends a `SM_MESSAGE_ID_DATA_READY` which is then processed in `...TaskExecuteStepDatalog`.

I noticed that when I read the corresponding data with the **hsdatalog_SDK** after a certain time there were corrupted data, resulting to the following check:
https://github.com/STMicroelectronics/fp-sns-datalog2/blob/bb0a1d7560b5c459908b112e091f42f284670259/Utilities/HSDPython_SDK/st_hsdatalog/st_hsdatalog/HSD/HSDatalog_v2.py#L600) 

I tried to find the root cause in the firmware, and to do this I supervised the time interval between each sample as follows:

```
case SM_MESSAGE_ID_DATA_READY:
{
  res = ...TaskSensorReadData(p_obj);
  if (!SYS_IS_ERROR_CODE(res))
  {
    // notify the listeners...
    double timestamp = report.sensorDataReadyMessage.fTimestamp;
    double delta_timestamp = timestamp - p_obj->prev_timestamp;
    
    if (delta_timestamp > 1.5f * (..._TASK_CFG_TIMER_PERIOD_MS / 1000.0f))
    {
      SYS_DEBUGF(SYS_DBG_LEVEL_WARNING, ("Time interval exceeded %.2f\r\n", delta_timestamp));
    }
  ...
  }
}
```

In this way I've been able to see that from time to time a 1 sec. interval is detected and now, what's interesting is that when this 1 sec. interval appears, the detection of a lower battery level as indicated in the previous message is also detected.

Could these two effects be related?

Do not hesitate to contact me if you need more information for your investigation.

Kind regards,
Ludwig

pradolis, 2024-02-16:
Hello @Ferag-AG 
STTS22H on STWIN.box should be similar to your sensor. By selecting 50Hz as odr it streams 1 data every 20ms and ReadData is driven by a 20ms-based timer.
With the patch I'm proposing, I'm not able to see the issue anymore in DATALOG2 official release.

Can you enable only the STTS22H + my proposed patch? 
By excluding your task, we can try to understand if there could be also a bug in the ReadData of your sensor.
I'm guessing: are there any blocking routines in you ReadData, maybe related to the sensor driver? Can you debug and measure the ReadData time execution?

Best regards
Simone

Ferag-AG, 2024-02-16:
Hi @pradolis 

In order to get trace of interval time recording I have implemented interval time check in the tasks STTS22H too.

When I activate STTS22H only as proposed above, neither data recording stops nor an excessive interval time is detected. But when I just use a part of the original sensors (see device_config.json) then I sometime gets time interval exceeded in the STTS22H task and when I try to read the data using **hsdatalog_SDK** corrupted data are detected.

`Sensor stts22h_temp: corrupted data at 39.88435325625 sec`

I have attached the corresponding recording if this can help.: [DL2_00006.zip](https://github.com/STMicroelectronics/fp-sns-datalog2/files/14311157/DL2_00006.zip)

So I suppose that you could be able to reproduce this on your setup too.

Kind regards,
Ludwig

pradolis, 2024-02-19:
Hello,
we found the (real) bug in the ADC configuration, shared between STWIN and STWIN.box (so, the issue is there also in STWIN example).
We are testing the solution and push it in the upcoming v2.1.0

Best regards
Simone

smellit, 2024-02-19:
Hi Simone

Great to now that you find the bug 👍🏻. What is the expected frame time for v2.1.0?

Kind regards,
Ludwig

Ferag-AG, 2024-02-19:
Hi Simone

Great to now that you find the bug 👍🏻. What is the expected frame time for v2.1.0?

Kind regards,
Ludwig

pradolis, 2024-03-12:
Hello @Ferag-AG,
v2.1.0 has been released on [st.com](https://www.st.com/en/embedded-software/fp-sns-datalog2.html) only.
This new version includes the fix on battery management, so I'm closing the issue.

Feel free to contact us again if the issue persists on your side.
Best regards

closed 2024-03-12 by pradolis

Ferag-AG, 2024-03-13:
Hi @pradolis

Thanks for the notice.

I tried to download the module from ST, but the link to the page after selecting the version doesn't work.
Why do you not publish the new version here?!

Regards

pradolis, 2024-03-20:
Hello @Ferag-AG 
v2.1.0 has just been pushed also on github.

Best regards
Simone

Ferag-AG, 2024-04-03:
Hi @pradolis 

I can confirm that this issue is fixed in v2.1.0 👍🏻 

Best regards
Ludwig