STMicroelectronics/fp-sns-datalog2 #13: Does not run on STEVAL-STWINKT1B

bitmorse, 2024-05-02:
**Setup**

 * STEVAL-STWINKT1B
 * CubeIDE 1.13.2

**Describe the bug**

Cloning the project and compiling the respective project (Projects/Applications/STM32L4R9ZI-STWIN) using CubeIDE works.
Flashing the project (ST Link V3) on to the board works, however the project does not appear to be running. 

- "dmesg" shows a "STM32 BOOTLOADER" attached instead of the HSD device
- LED not blinking
- No BLE service discoverable

Flashing the datalog1 firmware project using the same method also works and the firmware runs. dmesg shows the HSD device.

**How To Reproduce the bug**
1. Flash this repository's Projects/Applications/STM32L4R9ZI-STWIN project on a freshly bought STEVAL-STWINKT1B

pradolis, 2024-05-02:
Hello @bitmorse 
STWIN board has  a double-bank flash. Can you check if you are flashing/running the firmware on the proper bank?
[image]
To flash the code using one of the compatible IDE and debuggers, you need to make sure that the active flash bank is the first one. You can do this using the STM32CubeProgrammer:
[image]
This is the proper MCU setup. 
Check this and then let you know if it restarts working properly.

Similar procedure is described for STWIN.box and SensorTile.box PRO.
To avoid possible issue in future, we'll improve the related chapter in User Manual and Quick Start Guide also for STWIN.

bitmorse, 2024-05-04:
That fixes it! It works perfectly now. Thanks a lot!

closed 2024-05-04 by bitmorse