STMicroelectronics/fp-sns-datalog2 #15: cli_example not working on RPI Model B (32/64 bit Raspbian OS)

alejandrolabajos, 2024-07-09:
Hello, 
I am trying to run the cli_example script using my Raspberry Pi 4 model B, but I am facing different issues depending on the OS installed in the RPI.

_Some details before define how to reproduce the errors._
- STEVAL-STWINBX1 Firmware: FP-SNS-Datalog2 v2.1.1. Boot using Release binary file provided.
- Device: Raspberry Pi 4 Model B 8GB with a fresh boot.
- STM board is listed correctly using lsusb on RPI4.
[image]
- All utilities working fine on my host PC (Windows 11).

**Error using Raspberry Pi OS (bookworm) 32-bit or Raspberry Pi OS Legacy (Debian based) 32-bit.** 

Same error encountered with both 32-bit OS.

_My_ _configuration:_

[image]
[image]

_Workflow:_

I followed the README_Linux instructions located in cli_example folder (/Utilities/cli_example), but instead of using files of Linux folders i've used the files located in /raspberryPi_setup and /bin_raspberryPi4_32.

[image]

_Error message received:_

[image]

**Error using Raspberry Pi OS (bookworm) 64-bit.** 

This is also mentioned in the Issue #8 by the user @francus_dev, the workflow proposed by @pradolis is not working, I got the same issue.

_My configuration:_
[image]

_Error message received:_

` ./cli_example: error while loading shared libraries: libhs_datalog_v2.so: wrong ELF class: ELFCLASS32 `

Any help will be appreciated, since i am not an expert and i didn't have this type of difficulties running it on my Windows host PC. 
Thanks.

alejandrolabajos, 2024-07-10:
Marked as closed. 

**SOLVED:** Regarding the section "Error using Raspberry Pi OS (bookworm) 32-bit or Raspberry Pi OS Legacy (Debian based) 32-bit.", the problem appeared because I wasn't disconnecting the board before running setup.sh script located in /raspberryPi_setup. Executing the script with the board disconnected, and then connecting the board to RPI and running cli-example located in /bin_raspberryPi worked perfect.

**NOT SOLVED:** Regarding the section "Error using Raspberry Pi OS (bookworm) 64-bit." I did not found the cause for this issue.

closed 2024-07-10 by alejandrolabajos

pradolis, 2024-07-10:
Hello @alejandrolabajos 
Thanks for your feedbacks. If I well understood, the issue that still persists is the one already shared in #8 .
Am i right?

Best regards
Simone

alejandrolabajos, 2024-07-10:
Hi @pradolis 

You are totally right.

Thank you for your fast reply,
Alex