STMicroelectronics/fp-sns-datalog2 #16: Python HSDatalog_GUI not detecting STEVAL-STWINBX1

alejandrolabajos, 2024-07-11:
Hello,

I am trying to run hsdatalog_GUI.py interface on my Raspberry Pi 4 model B (64-bit). 

OS configuration:
[image]

I followed the instructions explained in README_Linux file, located at /Utilities/HSDPython_SDK:
[image]

Everything works fine during installation, including PySide6 package. 

The problem comes up when I try to run hsdpython_GUI.py, which is not detecting my STWINBX1 device.
[image]

The board is detected correctly using lsusb:
[image]

Output of terminal: 
[image]

Output Logs of GUI:
[image]

I also tried to use the library provided in /Utilities/cli_example/bin_raspberryPi4_64, but as reported in Issue #15 , seems there is some problem with this lib for aarch64 64-bit OS. 

- I am not able to use this GUI with RPI 32-bit OS since there is not support of PySide6.
- Is running perfectly on my Windows 11 host PC.

Any help will be appreciated.

pradolis, 2024-07-16:
Hello @alejandrolabajos 
I'm testing your setup in v2.2.0 and it is working. 
Pi4 has a limited RAM and USB capabilities, so we are still facing some video latency while logging many data.
Pi5 instead seems working better.

Remember that FP-SNS-DATALOG2 is not exposing datalog GUI natively in Raspberry PI installers. GUI is fully tested and intended for Windows and Linux PC only. That's why Raspberry installer setups automatically the "no_GUI" environment only. Furthermore, 32bit RPI OS is not supported.
To make "GUI" running, you must do some further manual steps (as you described), thus generating potential issues.

Is the issue systematic? Can you reboot the Raspberry once you face the issue and check how it works?

Best regards
Simone

alejandrolabajos, 2024-07-19:
Hi @pradolis 
 
yes, same error every time and same logs

pradolis, 2024-08-13:
Hello @alejandrolabajos 
No further news on the topic, so I'm closing the issue.
Feel free to reopen the issue if needed.

Best regards
Simone

closed 2024-08-13 by pradolis

alejandrolabajos, 2024-08-13:
Hi @pradolis, 

I just checked the version 2.2.0 with configuration and mlc setup provided at #19 and everything is working perfectly on my RPI4B (64-bit OS).

Thank you so much for your effort and support!!
Alex