STMicroelectronics/fp-sns-datalog2 #17: hsdatalog_GUI does not store the device configuration on the SD card (STWIN.box).

Ferag-AG, 2024-07-12:
Hi

Uses version 2.1.1.
Hardware: STWIN.box

In the HSDPython_SDK hsdatalog_GUI example, when I modify the device configuration and store it on both the computer and the device, the device_config.json file on the computer is updated correctly, but the device_config.json file on the SD card is not.

I hope you'll be able to check this, because if it doesn't work as expected, and the STWIN.box couldn't be configured as intended for measurement recording, this may lead to having to redo a whole set of measurements.

Thanks in advance.

pradolis, 2024-07-16:
Hello @Ferag-AG 
Added in pipe for v2.2.0, planned for end of July.

Best regards
Simone

pradolis, 2024-07-17:
Hello @Ferag-AG 
I confirm the bug on the FW side. Fixed in upcoming v2.2.0.
I will keep the issue open and notify you once v2.2.0 is online. Then, I'll close the issue.

Best regards
Simone

pradolis, 2024-07-29:
Hello @Ferag-AG 
I've just pushed the v2.2.0.

Best regards
Simone

closed 2024-07-29 by pradolis