STMicroelectronics/fp-sns-datalog2 #18: STWIN.box WiFi / FTP configuration

Ferag-AG, 2024-07-12:
Hi,

Does the WiFi and FTP connection work on the STWIN.box. I try to find out how to use it, but there is no documentation about this.

This function will be a very valuable improvement instead of having to remove the SD card every time after stand alone measurement.

Regards

pradolis, 2024-07-17:
Hello @Ferag-AG,
Have a look at this thread: [STEVAL-STWINBX1 datalogger - Cannot get WiFi to work ?](https://community.st.com/t5/edge-ai/steval-stwinbx1-datalogger-cannot-get-wifi-to-work/m-p/696094).
You can find the solution there.

Detailed instructions on wifi/ftp configuration are available in the [User Manual](https://www.st.com/resource/en/user_manual/um3106-getting-started-with-the-stm32cube-function-pack-for-high-speed-datalogging-and-ultrasound-processing-stmicroelectronics.pdf) (chapter 2.4 FTP Data Retrieving over Wi-Fi) and in the [Quick Start Guide](https://www.st.com/resource/en/product_presentation/fp-sns-datalog2-quick-start-guide.pdf) (section 2.1.5 FTP Data Retrieving over Wi-Fi)

Let me know if you are satisfied (so I can close the issue) or if you need further support.
Best regards
Simone

Ferag-AG, 2024-07-17:
Hi @pradolis 

Thanks for the clarification! I think this information will be sufficient. I can't test it right away but you can nevertheless close this issue.

Best regards
Ludwig

closed 2024-07-17 by pradolis