STMicroelectronics/fp-sns-datalog2 #19: Bus error loading device_configuration.json + MLC_decission_tree.ucf STEVAL-STWINBX1

alejandrolabajos, 2024-07-15:
I created a custom classificator .ucf using UnicoGUI SDK. 

The workflow i followed is:

Windows 11: 

1. Acquire data and create .ucf file with UnicoGUI. Working fine (except Unico data converter which is not working, conversion done with hsdatalog python based converter script)

2. Use hsdatalog_GUI to configure the sensors and load the MLC configuration file. Working and classifying perfectly.

3. Save device_configuration.json and MLC_configuration.ucf files for its use on final target device RPI4B.

Then moving to the final target device RPI4B: 

1. Not able to use hsdatalog_GUI (board not recognised, Issue #16 ).

2. Using cli_example C++ based interface with the next command throws a "Bus error" after "HSD_DLLv2: DeviceManager created" message. `./cli_example -t 10 -f device_configuration.json -u MLC_configuration.ucf`

pradolis, 2024-07-17:
Hello @alejandrolabajos 
As commented out in #16 , the setup seems working on my side.
Can you share both device_configuration.json and MLC_configuration.ucf you are using?

Best regards
Simone

alejandrolabajos, 2024-07-17:
Sure! the MLC file is basically a vibration classifier of a fan using different velocities, taking the input from ISM330DHCX Accelerometer and Gyro (as the MLC datasheet info says, both with ODR 104Hz, FS=2g on Acc and dps=125 on gyro). 
The device_config enable all sensors i need to use for datalogging, with ISM330DHCX parameters matching the previous config for creating MLC file.

[device&MLC_config.zip](https://github.com/user-attachments/files/16262729/device.MLC_config.zip)

Many thanks!

pradolis, 2024-07-19:
Hello @alejandrolabajos 
I replicate your setup and both C++ and Python GUI works on my side.
I guess the issue on your setup depends on a misconfiguration during the SDK setup (maybe you launched the 32 bit or linux setup instead of the 64 bit?).

v2.2.0 is fixing some troubles (both on firmware and software), so once I'll push the package I suggest you to update it.
Package will be closed soon and I should be able to push it in a couple of weeks.

Best regards
Simone

alejandrolabajos, 2024-07-19:
Hi @pradolis 

thank yofor u a lot for the support, i'll wait for v2.2.0 and try it again watching carefully which setup i launched. For the moment I will close this issue and I will come back with my feedback and reopen it if needed.

Many thanks, 
Alex

closed 2024-07-19 by alejandrolabajos