STMicroelectronics/fp-sns-datalog2 #22: Realtime Data Recording cannot be stopped properly

Ferag-AG, 2024-11-21:
Hi all

I use a Raspberry to record real-time data from a STWIN.box. I execute consecutive sequences of measurements, but when stopping the recording, the `stop_log` function sometimes gets stuck, and I've also noticed that a timeout in libusb occurs and the output console is filled up with:

```

```

fp-sns-datalog2 version used: 2.2.0

**Rasberry Infos**

```
==== Raspberry Pi Hardware and Software Summary ====
Model and Revision:
Revision	: d03115
Model		: Raspberry Pi 4 Model B Rev 1.5

OS Information:
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian

Kernel Version:
6.6.51+rpt-rpi-v8

Architecture (32-bit or 64-bit):
64-bit

Firmware Version:
Aug 30 2024 19:17:39 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)

Memory Info:
               total        used        free      shared  buff/cache   available
Mem:           7.6Gi       1.5Gi       5.2Gi        88Mi       1.1Gi       6.1Gi
Swap:          199Mi          0B       199Mi
arm=948M
gpu=76M
```

pradolis, 2024-11-21:
Hello @Ferag-AG 
I've just reproduced your setup and started a series of acquisitions. 
I've been doing a dozen of consecutive acquisitions for a couple of hours. As for now, I wasn't able to get your error.
Is the issue repetitive? Is there a specific setup or sequence of actions that you find out generating the error?

Best regards,
Simone

Ferag-AG, 2024-11-21:
Hi @pradolis 

The issue recurs repeatedly when measurements are in progress, sometimes after only a few recording sequences. I use two STWIN.boxes connected via USB with standard FW. The only thing that may be different is that I don't have a battery in the STWIN.boxes, so the setup is straightforward.

Regards,
Ludwig

Ferag-AG, 2024-12-10:
Hi @pradolis 

I just tried V2.3.0 of the SDK with corresponding FW on the STWIN.box and the problem stay the same. I also experienced the same behaviour using hsdatalog_GUI.py when stopping recording when two SWIN.boxes are connected by USB.

Regards,
Ludwig

akashintera, 2025-03-06:
I also have the same issue and the same error. This problem is with all the examples where the stop log runs consecutively.

pradolis, 2025-03-07:
Hello @akashintera,
The procedure related to file closing and opening depends also from the host side.
If the host is busy for any reason, or host resources are limited, maybe 1 second isn’t enough to close all the files.
In this case, DATALOG2 isn’t still ready for the next acquisition, thus causing the fault while trying to restart.

I see that _Idle Period_ is set to _1_ in your screenshot and I guess you are on a Linux-based machine (maybe a Raspberry?). 
Can you replicate your experiments by increasing the idle period?

Best regards,
Simone

akashintera, 2025-03-10:
Yeah I am running it on Raspberry Pi 4. 

I increased the idle period but the issue is the same. Can you suggest some recommendations?.  

Also I tried to recreate a python script to recreate the autologging mode but the issue is the same. 

hsd_link.stop_log(hsd_link_instance, 0) hangs the device at any random iteration. The error is "1Failed to reset the device"

"# Create HSD Link instance
hsd_link = HSDLink()
hsd_link_instance = hsd_link.create_hsd_link(dev_com_type='st_hsd', acquisition_folder='./test_folder')
print("Connected:", hsd_link_instance is not None)

# Global flags
is_logging = False
is_waiting_idle = False

# Constants
LOG_DURATION = 100  # Time to log (seconds)
IDLE_DURATION = 10  # Time to idle between cycles (seconds)
STOP_RETRY_DELAY = 2  # Delay between stop log retries (seconds)
MAX_STOP_RETRIES = 5  # Maximum retries to stop the log

def format_time(seconds):
    """Formats time as HH:MM:SS."""
    minutes, seconds = divmod(seconds, 60)
    hours, minutes = divmod(minutes, 60)
    return f"{hours:02d}:{minutes:02d}:{seconds:02d}"

def start_log():
    """Starts logging process."""
    global is_logging
    is_logging = True

    print("Starting log...")
    hsd_link.start_log(hsd_link_instance, 0)
    print("Logging in progress...")

    elapsed_time = LOG_DURATION
    while elapsed_time > 0:
        print(f"Automatic Mode ---> LOGGING: {format_time(elapsed_time)}")
        time.sleep(1)
        elapsed_time -= 1

def stop_log_attempt():
    """Attempts to stop the log with retries."""
    global is_logging, is_waiting_idle

    is_waiting_idle = True
    elapsed_time = LOG_DURATION

    while elapsed_time > 0:
        print(f"Automatic Mode ---> LOGGING: {format_time(elapsed_time)}")
        time.sleep(1)
        elapsed_time -= 1

    is_logging = False
    is_waiting_idle = False

    for attempt in range(1, MAX_STOP_RETRIES + 1):
        try:
            print(f"Stopping log... Attempt {attempt}")
            hsd_link.stop_log(hsd_link_instance, 0)
            print("Log stopped successfully")
            return  # Stop successful
        except Exception as e:
            print(f"Error stopping log: {e}")
            time.sleep(STOP_RETRY_DELAY)

    print("Failed to stop log after multiple attempts.")

def run_auto_mode():
    """Runs the automatic logging cycle indefinitely."""
    cycle = 1

    while True:
        print(f"\nCycle {cycle} starting...")

        # Wait before starting log
        wait_time = 5
        print(f"Waiting to start log... {wait_time} seconds")
        time.sleep(wait_time)

        # Start logging
        start_log()

        # Stop logging with retries
        stop_log_attempt()

        # Enter idle period
        print(f"Entering idle period for {IDLE_DURATION} seconds...")
        time.sleep(IDLE_DURATION)

        cycle += 1  # Increment cycle count

# Main execution
if __name__ == "__main__":
    if hsd_link_instance:
        print("Updating device configuration...")
        hsd_link.update_device(hsd_link_instance, 0, json_file_path)
        print("Device updated successfully.")

        print("Uploading UCF file...")
        hsd_link.upload_mlc_ucf_file(hsd_link_instance, 0, ucf_file_path)
        print("UCF uploaded successfully.")

        time.sleep(5)  # Allow stabilization

        # Start automatic logging cycle
        run_auto_mode()
"

pradolis, 2025-03-14:
Hello,
issue seems related to raspberry pi 4 USB management. It’s the same as in #26.
I'm not able to reproduce it in Windows neither in other Unix environments.

I'm closing this issue as duplicate and I'll update #26 only.

Best regards
Simone

closed 2025-03-14 by pradolis