STMicroelectronics/fp-sns-datalog2 #23: No changes for STM32U585AI-SensorTile.boxPro in v2.3.3

jankap, 2025-01-08:
Is it correct that no changes were made in v2.3.3 for the STM32U585AI-SensorTile.boxPro or are there copy & paste errors?

Thanks

pradolis, 2025-01-09:
Hello @jankap 
what are you referring to? There’s no v2.3.3 available. The last version published is the v2.3.0 and it differs from the previous v2.2.0.

Best regards
Simone

jankap, 2025-01-09:
Excuse me, I'm referring to 2.3.0.

There are a lot of changes to other projects/boards, but not for the SensorTile Pro. I'm just curious why the Bugfixes don't apply to that board, too.

jankap, 2025-01-09:
Dear @pradolis I must apologize. Downloaded the release.zip and just compared it in my local SCM and apparently, I've screwed it up. 

I Can see the changes for SensorTile.boxPro now locally.

Closing this.