STMicroelectronics/fp-sns-datalog2 #25: Debug frozen firmware

jankap, 2025-01-17:
We're using the latest version of DATALOG2 with automode enabled. We want to measure multiple days and purchase big SD cards (64 GB, formatted to 32 GB FAT32). We're using the sensor tile box pro.
Unfortunately, the firmware acquisition stops randomly and instead of blinking (fast or slow, depending if device is recording or idle) LED, the LED is just on. No reaction to any buttons anymore. 
Also, the power consumption increases from 50-80 mA to static 155 mA in that error use.

It happened twice now, first time after 11 h, second time after just 3 h.

The boards are not connected to any PC, but USB powered by a charging adapter.

What does the constantly enabled LED indicate? Any ideas why there's a higher current consumption?

Thanks a lot,
Jan

jankap, 2025-01-20:
Update: I made various trials with different partition sizes. The firmware crashes everytime when the partition size is > 8 GB. Is there a problem in the underlying FAT32 lib or where can I start to investigate? I'm on some pressure to get this solved here.

PS, the generated `device_config.json` always shows `64` as `totalStorage`, no matter what actual size the partition has:

[image]

The problem also occurs when the board is connected to the computer via USB.

pradolis, 2025-01-20:
Hello @jankap 

> I fully get that. The interesting part is that if I partition the same SD card's first partition to have 8 GB, the FW does not crash. When it's 16, or 32 GB it crashes. Same configuration otherwise, therefore I do not expect buffering issues? How does the partition size relate to buffering or writing speed?
> 
> Ps, the card is also a class 10 sdhc from Sandisk. Similar to the one you tested.

Maybe I misunderstood your question: can you share the whole list of parameters you’re using for SD card formatting?
I'm simply formatting my SD in Windows environment with the very base tool from Windows, selecting properly the allocation table dimension only.
[image]
Once you’re partitioning the SD card are you modifying the allocation table also?

Full details on SD card issues are in the already cited paragraph of the User Manual. Have you already had a look into?

> Are 70 mA reasonable for just one LED? However, thanks for explaining the hard fault indicator. I'll try to attach a debugger and trigger it. Any recommendations what to look for?

Let me double check internally, I'll come back soon on the specific topic.

> No, streaming is off. USB is only for power.

So, the issue is on the SD card side only. Selecting USB or battery as power supply doesn't influence the SD card at all.

Best regards
Simone

jankap, 2025-01-20:
> Hello [@jankap](https://github.com/jankap)
>
>
> Maybe I misunderstood your question: can you share the whole list of parameters you’re using for SD card formatting? I'm simply formatting my SD in [...]
Sure. I have one 64 GB SD card, ‎SDSQUAB-064G-GN6FA, micro SDXC, e.g. https://shop.sandisk.com/de-de/products/memory-cards/microsd-cards/sandisk-ultra-uhs-i-chromebook-microsd?sku=SDSQUAB-064G-GN6FA. 

I partitioned that card using the Windows-built-in `disk manager`, see https://learn.microsoft.com/en-us/windows-server/storage/disk-management/overview-of-disk-management. 

I create the partition and format it as FAT32, 32 k allocation table.

[image]

[image]

Since FAT32 on Windows seems to work only up to 32 GB, this is my maximum test size, although the card is 64 GB in size and comes with extFAT file system per default. I deleted that partition and created one FAT32 partition at the very beginning of the SD card.

In the screenshot it's 32 GB, but I tested also 16 GB, and 8 GB. 

32 GB and 16 GB failed, i.e., the LED just kept on after a while (between an hour and 12 h). I tested that on different sensor tile box pro hardware with the original and self-compiled firmware in v2.3.0, I tested Debug and Release builds.

I could not yet produce an error with an 8 GB partition.

> Full details on SD card issues are in the already cited paragraph of the User Manual. Have you already had a look into?
> 

I have had a look, yes. The only thing I've not tested yet is the `HSD_USE_DUMMY_DATA` macro, I will try that.

> > Are 70 mA reasonable for just one LED? However, thanks for explaining the hard fault indicator. I'll try to attach a debugger and trigger it. Any recommendations what to look for?
> 
> Let me double check internally, I'll come back soon on the specific topic.
> 
Sure, thanks!
> > No, streaming is off. USB is only for power.
> 
> So, the issue is on the SD card side only. Selecting USB or battery as power supply doesn't influence the SD card at all.
> 
Yes! It's only this one particular SD hard. Please note that I used h2testw, see https://www.heise.de/download/product/h2testw-50539 (sorry, German only) to validate that the 32 GB partition is properly working and that I have no SD card clone or similar.

> Best regards Simone

Best regards,
Jan

PS, I also ordered exactly the card quoted in the documentation, should arrive soon.

pradolis, 2025-01-21:
Hello @jankap 
I've just found a 128GB SD from Sandisk. I formatted it as you described and started a recording session with automode to replicate your setup.
You said the firmware is blocking after hours... I'll make some experiments and come back in a couple of days.

Best regards
Simone

jankap, 2025-01-21:
Hi @pradolis 
I had a test overnight. Same config, same firmware, two identical STEVAL-MKBOXPRO with external ISM330.
One with the 64 GB SD-card, formatted to 32 GB, FAT32, 32k cluster.
The other sensor had a crappy, multiple years old no-name 4 GB card.

The 64 GB SD-card crashed over night, the other still runs.

I'll continue testing with other 64 GB cards until the 32 GB cards arrives.

> I've just found a 128GB SD from Sandisk. I formatted it as you described and started a recording session with automode to replicate your setup.

Wonderful, thank you!

PS, the config I'm using is this (remove the `_highspeed` when copied to card), maybe you could test with that config?
[device_config_highspeed.json](https://github.com/user-attachments/files/18488696/device_config_highspeed.json)

I think there's a dependency on the sampling speed. The config maxes out all sensors (this is not the config I'll use in production, this is just to trigger the error quicker).

Thanks!

jankap, 2025-01-21:
Update on my test using a *different* 64 GB SD-Card, but same type: Error occurred after ~2 h.

So, this problem does not only appear with one specific (potentially broken) SD-Card but with at least two SD-Cards of the same type.

I'll now re-partition the card to have 16 GB, (currently 32 GB) and redo the test.

pradolis, 2025-01-27:
Hello @jankap 
my 128GB Sandisk is an SDHX type and I got your same malfunctioning.
In the past, we've never tested using SD card bigger than 32 GB, and all of them were SDHC originally formatted FAT32.
I'm sorry, but I still don't have a definitive solution. 
Considering the experiments we did, I can only state that we're expecting DATALOG2 to work properly with SDHC FAT32 with certainty.

The issue is related to the memory access time.
If an SD card is designed to support an average writing throughput that may be even far above 6 Mbit/s, sometimes they might also present a time-varying latency with hundreds of milliseconds peaks. 
So, a “good” SD card having a very high baud rate in average, can have also maybe a single outlier, a long latency, that impacts definitively firmware performance.

There’s something weird (and still not understood) once using SDHX. 
Upcoming v3.0.0 will still support SDHC FAT32 SD cards only.
But I'm taking notes about all the test that we're doing, trying to find a new approach for supporting SDHX also in the future.

Best regards
Simone

jankap, 2025-01-27:
Dear @pradolis 

thank you so much for your investigations and confirming the issue with bigger (SDHX) cards. I'll redo the tests this week once the 32 GB SDHC cards have arrived :)

> But I'm taking notes about all the test that we're doing, trying to find a new approach for supporting SDHX also in the future.

This is highly appreciated! I'd suggest to add a comment to the README to highlight limited SDHX compatibility.

pradolis, 2025-01-29:
Hello @jankap 
If the system is battery-powered, the firmware checks automatically battery level and performs a sanity check.
If the level is under a threshold, it automatically stops to avoid data loss.
If instead is USB-powered, this power supply sanity check is not anymore needed and (obviously) performed.

Best regards
Simone

jankap, 2025-02-25:
Thanks for all your comments @pradolis I'll close the issue for now :)

closed 2025-02-25 by jankap