STMicroelectronics/fp-sns-datalog2 #26: Cli_example failing at stop when executed repeatedly

alejandrolabajos, 2025-03-10:
Hello @pradolis 

I am trying to develop an autonomous vehicle based FSM using STEVAL-STWINBX1 with DATALOG2 (V3.0.0) to acquire data from almost all its sensors. 

The first execution with cli_example works fine, but after the second execution, after stopping the datalog with keyboard (ESC) it gets hanged waiting for something to finish the execution. Working fine it takes 1-2 seconds to finish, when this thing occurs, waits hanged for about 15-20 sec and then starts throwing the same error in the terminal.
I am using a RaspberryPi 4 model B (tried with both, 32-bit and 64-bit OS).

Here is the explanation step by step.

RPi4B info:

[image]

I followed the instructions explained in README_raspberrypi.md file, located at /Utilities/cli_example:

[image]

Everything works fine during ./setup script execution. I close this terminal and reopen a new one from /cli_example/ folder following the instructions.

I ran ./bin_raspberryPi_32/cli_example and as you can see is starting  and stopping perfectly:

[image]

[image]

Now I wait for arroud 10 seconds and, in the same console, I run again ./bin_raspberryPi_32/cli_example. 
Works fine during datalogging but when I stop the the program hangs before ending execution:

[image]

And after 10-15 seconds being hanged, starts throwing errors:

[image]

I don't know what's happening but I think it's somehow related to error #22 because the output error is the same (not the format than @Ferrag-AG but the same as @akashintera)

Many thanks!
Alex

alejandrolabajos, 2025-03-14:
Hello @pradolis 

Yes, also from my side the library seems working fine on Windows.

Thank you for keeping me updated, I really apreciate it.

Regards,
Alex

pradolis, 2025-03-19:
Hello,
we fixed the issue. [Here](https://github.com/user-attachments/files/19337598/bin_raspberryPi_64bit.zip) the updated cli_example + libhs_datalog_v2 for Raspberry 64bit.
Can you try it out?
If it is working properly also on your side, we'll push it on the repo.

Best regards
Simone

alejandrolabajos, 2025-03-19:
Hello, 

In our RPi, when we first ran the new cli_example from the link above we've got this error regarding undefined symbols.

[image]

But worked after we exported the library:

[image]

First execution taking data for about 12 sec finalyzed fine and fast after stopping it. Second execution too. 
But at third execution it got stucked after stopping it (not throwing that error message about libusb in the whole terminal but still frozen).  After a while, the terminal works again but then no detect the device anymore untill rpi is rebooted:

[image]

Note: We also tried to run cli_example with Linux-based OS from the device Nvidia AGX Orin 64GB, but seems it can't be run here probably because of architecture incompatibilites (Error: bash cannot execute binary file).

akashintera, 2025-03-19:
I get this error  

[image]

totino5, 2025-04-03:
Hi @pradolis, I noticed the same problem using stdatalog-pysdk, so now I'm stucked on a work project where I use both Raspberry PI Model 4 and STwin.box. Do you have any idea when the new version that fixes this bug might be released? 

Thank you!

pradolis, 2025-04-04:
Hi all,
we finally solved a couple of bugs and installation misconfigurations strictly related to Raspberry Pi 4.
First, we improved libhs_datalog_v2.so.
Then, we found out that the USB communication issue is solved by updating the libusb version. Natively, Raspbian distribution installs an old libusb version (1.0.9 I guess). By manually installing more recent versions (1.0.27 or 1.0.28, the latest published few days ago) the issue disappears.
Furthermore, we added the support for MacOS also.

We are finally closing the internal validation in few days. Next week we'll share the updated libraries and python modules.

Thanks for you patience
Best regards
Simone

totino5, 2025-04-07:
Hi @pradolis, thank you for your update. 

In my Raspberry PI configuration, I have the libusb 1.0.26 installed. The stdatalog gives the libusb error with this version so I tried to update to a newer one (1.0.28) following the github repo instructions. By launching this command after installation:

``` dpkg -l | grep libusb ```

I see this output:

``` 
ii  libgusb2:arm64                       0.3.10-1                            arm64        GLib wrapper around libusb1
ii  libusb-1.0-0:arm64                   2:1.0.26-1                          arm64        userspace USB programming library
ii  libusbmuxd6:arm64                    2.0.2-3                             arm64        USB multiplexor daemon for iPhone and iPod Touch devices - library
 ```

So the newer version isn't installed. Do you have some suggestions on how to update the library? Plus, will the USB communication issue be fixed with the next version of the libhs_datalog? 

Thank you!

pradolis, 2025-04-14:
Hi all,
@totino5: updated [STDATALOG-PYSDK](https://github.com/STMicroelectronics/stdatalog-pysdk) has been pushed few days ago. I've just added a couple of comments related to the instruction guide.

@alejandrolabajos @akashintera: I'm pushing the same DLL used in the python SDK and updated C++ cli_example and installers

Is this solving your issues? Can you provide me a feedback?
If I'll not receive any news in this week, I'll consider the issue solved, so I'll close it.

Best regards
Simone

pradolis, 2025-04-14:
Hello @totino5 
if git commands are failing, it means you're not updating the sdk.
So, I can suppose you're still in trouble and all the rest of your comment are still related to an older instance of the sdk.
Are you sure git and access to github working properly on your machine?

Best regards
Simone

totino5, 2025-04-14:
Hi @pradolis,

Sorry for my bad explanations. I was meaning that I tried to update the sdk with git and also tried to re-install the whole sdk by formatting the SD card on Raspberry following your instructions on the repo, but with both method the code execution gives me the errors above. I also re-tried a few minutes ago but I have the same problems.

Pretty sure that the sdk installed on my Raspberry is 1.1.0 because during the installation, log shows that the library is being uptdated to the new version. 

I tried on Windows machine and everything works fine.

P.S: I found out that you are italian like me, so if you want I can give you a better explanation of the problem in italian.

Thank you

alejandrolabajos, 2025-04-15:
Hello @pradolis 

Installation of new `setup.sh` worked fine with no errors, but when I run `cli_example` I get this error:

[image]

Checking this library you can see that there is no version x.x.29, which is the one giving error:

[image]

Should I need to install something else or some change is needed inside installation script?

Thank you,
Alex

pradolis, 2025-04-15:
Hello @totino5 

> [...] 
> Unfortunally, I have another issue. When starting and stopping multiple consecutive acquisitions via the SensorAcquisitionThread (following "nb_stdatalog_communication.ipynb"), the program crashes after a number of acquisitions (it's variable but it will crashes) with the following log:
> 
> [image]

Is the led blinking? I guess no, so the firmware crashed and consequently USB communication is failing definitively.
Can you remember me which board, firmware and version you are using?

> [...]
> Then, when trying to re-launch my program, the acquisition doesn't start at all, showing this:
> 
> [image]

The board and the USB are already dead. You can't restart anymore. If you are in this situation, there’s no other way than reboot the board and the USB on your RPI.

totino5, 2025-04-15:
Hello @pradolis,

Yes, both green and red led are blinking, even after the program crashes.
 
I am using STEVAL-STWINBX1 flashed with the "fp-sns-datalog2" firmware version 3.0.0.

totino5, 2025-04-18:
Hi @pradolis, any news about the issues? Have you been able to reproduce? 

If you want more informations, we can also talk to each other in private so you can have a clear overview of the situation.

totino5, 2025-05-07:
Hi @pradolis, are there any news about the issue? 

I use the sensor in a work project with my company and we need updates because we want to retrieve some data on a vehicle and then send them to the cloud. We have a Raspberry 4 for the development but we are looking to buy a Raspberry 5 for the production environment. Do you know if the library works properly with RPi 5?

alejandrolabajos, 2025-08-27:
Hello @pradolis 

Any update on this issue? Do you have internally some roadmap for releasing a new version? We still need it for a project and would be nice to skip this error.

Thank you so much!
Alex

akashintera, 2025-11-13:
I made it work by connecting the STWIN using an fast charging USB hub.

pradolis, 2025-12-11:
Let me add a couple of further comments on the topic.
I've always used both RPI4 and 5 with an original power supply (15W and 27W respectively). I'm guessing, but if USB isn’t working properly on the RPI side due to some power issues, maybe that's why I've never got your issues anymore. This can't be controlled by both cli_example either PYSDK.
Furthermore, specifically for STWIN board (STEVAL-STWINKT1B), thanks to the contribution of a customer on community ([link](https://community.st.com/t5/stm32-mcus-boards-and-hardware/stwinkt1b-pysdk-can-no-longer-connect/m-p/857519)), I found out a bug in DATALOG2 firmware. See the link for further details; the upcoming v3.2.0 DATALOG2 firmware will solve the issue.

Now, I'm closing the issue. Both FP-SNS-DATALOG2 and STDATALOG-PYSDK updated releases are coming soon.
If you need further support or new suggestions, open new issues in the related repositories.